Misfittech/nano_stepper #2: compilation with (for) MECHADUINO_HARDWARE is broken after hash 684f9ac4

swk, 2016-12-20:
compiling 684f9ac4 for mechaduino works quite well with Arduino 1.6.13 IDE, Later commits fail. (I have not tracked down the specific commit yet.

Only mod to the code is to uncomment the define for MECHADUINO_HARDWARE

Error message 
In file included from sketch/nzs.h:11:0,
                 from sketch/nzs.cpp:8:
sketch/board.h:83:2: warning: #warning "Compiling source for Mechaduino NOT NZS" [-Wcpp]
 #warning "Compiling source for Mechaduino NOT NZS"
  ^
sketch/nzs.cpp: In function 'int menuTestCal(int, char**)':
nzs.cpp:39: error: 'PIN_SW3' was not declared in this scope
  while(digitalRead(PIN_SW3)==1)
                    ^
nzs.cpp:43: error: 'PIN_SW3' was not declared in this scope
  while(digitalRead(PIN_SW3)==0)
                    ^
sketch/nzs.cpp: At global scope:
sketch/nzs.cpp:534:9: warning: #pragma GCC target is not supported for this machine [-Wpragmas]
 #pragma GCC pop_options
         ^
exit status 1
'PIN_SW3' was not declared in this scope

swk, 2016-12-20:
Tracked this down to commit  9639027d (the most recent commit probably a missed ifdef heh)

Misfittech, 2016-12-20:
Thanks I have merged your changes into the code.

closed 2016-12-20 by Misfittech