Misfittech/nano_stepper #5: ctrlmode command not working

Fetzge, 2017-01-08:
The ctrlmode command does not set the controller mode correctly.
If I use e.g. ctrlmode 2 sysparams shows 108 for controllerMode and I see undefined symbols on the display (see attached file).
[image]

Setting the controllermode via sysparams is working correctly.

Misfittech, 2017-01-11:
This should be fixed in the latest firmware release of FW0.09  The sysparms are broken out into individual commands,

closed 2017-01-11 by Misfittech

Fetzge, 2017-01-11:
But I already used FW0.09. I removed the comments for sysparams, otherwise i wont be able to set the controller mode at all.
[image]
This is how the commandline looks like.

Misfittech, 2017-01-11:
I will check in just a bit that controller mode 3 is functional, it might have broke in the build as I normally use the simple mode.

Misfittech, 2017-01-11:
I have confirmed the bug and I am working on solution now.

Misfittech, 2017-01-11:
I have pushed a fix (FW 0.10)  for the ctrlmode command which fixes bug of not being able to set the control mode using the command.