Misfittech/nano_stepper #6: Connecting to Smoothie

pristinesource, 2017-01-18:
I just confirmed that I cannot use Smoothie's 3.3v logic directly on the level-shifted pins (step (0), dir (1) and error (10)).
I tried with Vin not connected (using usb for power) and with Vin connected to a 3.3v source from smoothie (usb not connected).
With both setups, the screen would come on and the firmware would load, but the motor would not step when signaled from smoothie.
Maybe I am doing something wrong, any suggestions?

Alternatively, I did test with modified firmware:
    - Use D3 (3) for STEP, TX (30) for Dir and RX (31) for Err/Ena.
    - Disabled Serial5 in NZS::begin.
    - Connected Vin to 3.3v.
Then it works great with my Smoothieboard exactly as expected.

I would prefer not to have to modify the firmware each time I update it.
Do you have any suggestions on what I did to hook it up incorrectly?
Anything I can test or retry? I would be happy to test anything and share my results.

Referencing comments from: https://github.com/Misfittech/nano_stepper/pull/1

kb-, 2017-01-21:
Are these pins configured as pull downs on your Smoothie?

pristinesource, 2017-01-22:
No, I have them set as regular output pins. Should I try them as pull down?  I will post my relevant smoothie config here shortly.  Right now it is set to use the default config for the x-y-z pins from smoothie's example config.  Their example config works with the onboard stepper controllers, and works great when I bypass the level shifters on the nano zero board.  I will try them as pull down here is just a little bit and post my results here.

Thank you for helping me! 😄

kb-, 2017-01-22:
Here's what I understand: You can see on the NZS's schematics that sending 3.3V to the STEP, DIR or ERROR pin wouldn't do anything. They're already pulled up by VIN through the 10k resistor.

The appropriate setting is apparently open-drain (it's ground when closed, floating when open), so the BSS138 would see 0Vwhen closed and 3.3V when open. Same for D0, D1 and D10 (BSS138 is wired as a level shifter) [http://smoothieware.org/general-appendixes#external-drivers](url)

coolio986, 2017-05-19:
For anyone still watching this issue, there is a firmware update that attempts to fix the issue with logic shifting from smoothie to NZS. The firmware version that supports this is 0.17. The firmware had pulldown for dir/step pins after logic shifting and if you have any noise whatsoever then the steppers would miss incoming steps. The 3.3v logic on the  shifters causes the voltage to drop to 2.0v. Considering that the minimum logic high is 1.815v if you have any ripple on the dir/step pins it could cause it to miss counts.

The pulldowns were removed in favor of floating inputs so that smoothie / ramps can drive the pins directly, effectively aiding in less noise on the pins.

pristinesource, 2017-06-16:
I am still having no luck with my Smoothie using the regular STEP, DIR, and ERROR pins.  I have tried the following:

- All testing using v0.19 FW with line commented out in boards.h to make it for the old hardware.
- Tried normal pin configuration for output (no pull up/down), had absolutely no movement
- Tried open-drain (the "o"option on the Smoothie config) as suggested by @kb- , this almost worked, I got some movement out of it, but it just twitched and didn't really move.
- Moved connections to use D3 (STEP), TX (DIR) and RX (EN) pins (modified firmware as in OP), set smoothie pins to normal output, and got perfect motion

I am very happy to try anything you want, I can take pictures/videos of the results and configurations.  I really just want it to work also.

In the mean time I will continue to experiment on my own to see if maybe I can figure something out.

coolio986, 2017-06-18:
Just about the only thing I can think of in this case is using an O-Scope. My above findings were based upon the o-scope readings. I would suspect that the transition from logic 0 to 1 (0v - at least 1.815v) isn't occurring even though you have tried open drain. My smoothie outputs an exact 3.3v which translates into roughly 2.0v on the NZS after shifting. 

You could try monitoring the direction pin using a multimeter and see if it changes voltage from 0v to 2.0v when you move that axis back and fourth. 

The only issue with measuring the 2.0v is you'd have to measure it after the logic shifter on the PCB and that can be very difficult since all those components are SMD.

Another test you could try is bonding the grounds together between the logic ground on the NZS, and the ground on the smoothie. If they are not at the same ground plane you could be having a ground loop feedback issue. 

If you cannot ground them together, then measure the voltage difference between the ground on the NZS and the smoothie. If the voltage difference is larger than 100mV that could be causing the issue also.

coolio986, 2019-12-04:
@pristinesource, Any updates on this?

closed 2019-12-17 by coolio986