Misfittech/nano_stepper #31: compile error 'Serial5' was not declared in this scope

Hellfish78, 2019-06-01:
Hi, i'm trying to compile the 0.39 but get this errors:

sketch\commands.cpp: In function 'uint8_t kbhit_hw()':

commands.cpp:1521:9: error: 'Serial5' was not declared in this scope

  return Serial5.available();

         ^

sketch\commands.cpp: In function 'uint8_t getChar_hw()':

commands.cpp:1526:9: error: 'Serial5' was not declared in this scope

  return Serial5.read();

         ^

sketch\commands.cpp: In function 'uint8_t putch_hw(char)':

commands.cpp:1530:9: error: 'Serial5' was not declared in this scope

  return Serial5.write((uint8_t)data);

         ^

exit status 1
sketch\nzs.cpp: In member function 'void NZS::begin()':

nzs.cpp:637:2: error: 'Serial5' was not declared in this scope

  Serial5.begin(SERIAL_BAUD);

  ^

'Serial5' was not declared in this scope

tarchive, 2019-07-09:
You are missing the Nano Zero board where the Serial5 definition is defined. Either it is not installed or you do not have it selected. 

To install, follow the install guide here. [http://misfittech.net/blog/arduino-package-install/](url)

Once installed "Nano Zero (Native USB Port)" can now be selected from board menu and compilation error Serial5 will be resolved

coolio986, 2019-12-03:
@Hellfish78 has this been resolved?

coolio986, 2019-12-04:
Changes committed 98510ac

coolio986, 2019-12-04:
@trampas, can you close this issue?

closed 2019-12-04 by Misfittech

DanTheHero, 2020-02-07:
Please i need to solve this issue

coolio986, 2020-02-08:
This issue is solved. If you're still having an issue please fill out a bug report.