Misfittech/nano_stepper #33: narrowing conversion error

hg42, 2019-06-21:
when compiling on Arduino IDE on Linux I got this conversion error:

```
eeprom.cpp:67:90: error: narrowing conversion of '4095' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
 __attribute__((__aligned__(FLASH_ROW_SIZE))) const uint8_t NVM_eeprom[EEPROM_SIZE]={0xFFF};
                                                                                          ^
exit status 1
narrowing conversion of '4095' from 'int' to 'uint8_t {aka unsigned char}' inside { } [-Wnarrowing]
```

I guess it is a typo...should it be 0xFF?

Or may be I don't understand the special purpose of initializing an uint8_t with 0xFFF :-)

coolio986, 2019-12-03:
@trampas Can you commit these changes? There are no conflicts and it appears that the nvm size is being set to all F's. I don't see any issue with this change

trampas, 2019-12-03:
There are some other bugs I have found, let me see if I can do a new
release.

Specifically I have found issues with the NVM as that the "={0xFFF}" only
sets the first value not the entire memory.  This has been shown to cause
some issues, depending on how the code is compiled.

Trampas

coolio986, 2019-12-04:
Changes committed c263fbb

coolio986, 2019-12-04:
@trampas, can you close this issue?

closed 2019-12-04 by Misfittech