Misfittech/nano_stepper #37: AS5047 Error on Mechaduino

hg42, 2019-06-21:
I got two Mechaduino clones for free, so I use them to check if I want to use Servo Steppers.

I assume it's not a priority for you, but I see Mechaduino is supported in some way.
So I wanted to give it a try and may be give something back (also see my PRs).

NZS has some interesting features, so I want to try some of them, and eventually switch to SmartStepper hardware if it fits for my purposes.

After ironing out some compile problems, I am now getting a stream of AS5047 Errors when trying NZS with two 
```
AS5047 Error
AS5047 Error
AS5047 Error
...
```

If I enable the LOG messages I get a repeating sequence of these:

```
...
AS5047 Error
stepper_controller.cpp  309: move
stepper_controller.cpp  314: sample encoder
ERROR:    as5047d.cpp  198: read command 0xFFFF failed
AS5047 Error
stepper_controller.cpp  317: Angles 0 0
stepper_controller.cpp  329: Angles 0 0
stepper_controller.cpp  348: angle delta 0 0 (0 0)
ERROR:    stepper_controller.cpp  634: Motor may not have power
stepper_controller.cpp  130: Home pin -1
stepper_controller.cpp  586: start up encoder
as5047d.cpp  110: csPin is 16
as5047d.cpp  115: Begin AS5047D...
as5047d.cpp  134: AS5047D diag data is 0x0000
ERROR:    as5047d.cpp  198: read command 0xFFFC failed
ERROR:    as5047d.cpp  198: read command 0x7FFD failed
as5047d.cpp  153: AS5048A diag data is 0xFFFF
ERROR:    as5047d.cpp  241: AS5047D problem
stepper_controller.cpp  592: cal table init
calibration.cpp  346: calibration table status is: 0
ERROR:    as5047d.cpp  241: AS5047D problem
ERROR:    as5047d.cpp  241: AS5047D problem
ERROR:    calibration.cpp  197: WE did some thing wrong
WARNING:  stepper_controller.cpp  596: start up encoder 63728
stepper_controller.cpp  598: start stepper driver
stepper_controller.cpp  630: measuring step size
stepper_controller.cpp  302: reset motor
ERROR:    as5047d.cpp  241: AS5047D problem
ERROR:    as5047d.cpp  241: AS5047D problem
ERROR:    calibration.cpp  197: WE did some thing wrong
stepper_controller.cpp  305: sample encoder
ERROR:    as5047d.cpp  198: read command 0xFFFF failed
AS5047 Error
...
```

This is with motor power off, because it tends to get a little hot, but enabling the power doesn't change the messages that much. I am sure it's not the cause of those messages.

May be I am doing something weird, but after looking into it from different sides I don't see what could be wrong.

First question first: should NZS work with Mechaduino hardware?

Also: how is it supposed to be configured?
I ask, because enabling MECHADUINO_HARDWARE wasn't enough.
At least I had to disable NEMA17_SMART_STEPPER_3_21_2017 and USE_STEP_DIR_SERIAL ...
So may be I am missing more of this kind. Though, looking into the sources didn't give me any hint for more necessary changes.

Some conditions and observations:

* I am using two Mechaduinos from the same manufacturer (currently clones)
* both give me similar results
* both work so far (no real application yet) with original Mechaduino firmware and the version from Kai Wolter
* I already tried some things
  * reducing the SPI frequency to 400000 (from the uncommented parameters)
  * increasing a loop count (from older source) and delays in AS5047 read functions
  * reducing NZS_CONTROL_LOOP_HZ to 3000 (just in case :-) )
  * I also disabled the AS5047 error checking (because it seems to be ignored in KaiWolt1990) but the errors were resulting in bad values
* I also tried an older version (when Mechaduino was mentioned last, commit 3bb24719b232861286abb50c6f29945f56d1503c)

Any ideas?
Is there anyone using NZS on Mechaduino?

hg42, 2019-07-04:
that seems to be caused by compiling as Arduino M0 (or also M0 Pro, both variant mzero, because the boot loader is 16k and the device calls itself Arduino M0).
Arduino pin numbers are different in several ways (SPI, CS..., also D2 D4 are swapped, which is very bad for the AS4954 and motor [getting hot], because VREF1 is always zero).

hg42, 2019-07-07:
I managed to create a board definition to compile as Arduino Zero, but with a 16k bootloader and upload like the Arduino M0 (Pro) with avrdude.
This works so far. If anyone needs this, please send an email...

closed 2019-07-07 by hg42

beniroquai, 2023-03-15:
HEy @hg42 I guess I have the same issue. It's very long time ago, but did you have a solution available by the time?

hg42, 2023-03-15:
it's a long time ago and I forgot all the details...

I fixed this by changing some configuration lines in board.h, but I am not sure, how the result was working at the end.
At that time I experimented with several software solutions.

see my fork hg42/nano_stepper
branch mechaduino_clone (seems I forgot to push it at that time, so I pushed it now)
commit 0155c299d395021ce07612a53cd81ed23268154e contains the changed board.h

My final conclusion was, that the frequency is too low for my purposes.
Note, my first requirement was to have a quiet motor (like with TMC chips where I came from), and only the second goal was protection against skipping. If your priorities are different, you might come to a different conclusion.

beniroquai, 2023-03-15:
Perfect! Thanks a lot! I'll give it a try now.

> Message ID: ***@***.***>
>