Misfittech/nano_stepper #39: update to the readme?

theholyduck, 2019-07-25:
I recently came across this project and while reading the readme. i decided to look up the servo42. it seems a bit weird to open a issue over this but i think they since have released the source code and hardware designs for their version of this project. 
https://github.com/makerbase-mks/MKS-SERVO42A 

i totally understand not wanting to support a Chinese low cost clone of your project. but if all that was stopping you was lack of documentation. i think that now exists.

npgoalkeeper, 2019-08-09:
Side note, the Servo42a github page no longer gives credit to this project or the mechaduino. They just say it was developed by makerbase on their aliexpress pages.

coolio986, 2019-12-03:
There is now support for the mks servo42a. @trampas can you close this issue?

Misfittech, 2019-12-03:
supports servo42a

closed 2019-12-03 by Misfittech