Misfittech/nano_stepper #40: Duet Board Support

vmario89, 2019-11-20:
Dear Misfit team,
i've got a question belonging to the manual you provide. At the moment i plan to wire up Smart Stepper to a Duet 2 Ethernet Board. In manual is written:
[image]

As i don't find any recent information i just wanted to ask you directly by creating this issue.
Can you provide some support information please?

kind regards, Mario

coolio986, 2019-12-03:
Since the Duet creates additional pulses on the enable line, it is possible that the smart stepper may not produce movement if the pulse is not seen correctly. This could result in a loss of print quality or uncontrolled behavior. What can be done as a workaround is to set the enable pin high with a external pullup between 5v and enable with a 1k resistor. This will keep the smart stepper always enabled.

vmario89, 2019-12-18:
Hi, i just came up by thinking about wiring to the 50 pin JTAG connector because this is prepared to provide additional external steppers (STEP, DIR, ENABLE, VIN, GND are available):

[image]

Do i oversee something? I think this way the wiring of Smart Stepper should be possible to do directly without any soldering of pullup resistors on the "External Test Points" of the pre-soldered TMC drivers, right?

coolio986, 2020-01-02:
@vmario89 No soldering is needed to force the enable high. The resistor can go between enable and 5v on the smart stepper only via the terminal board.

coolio986, 2020-02-05:
@vmario89, is this working now?

vmario89, 2020-03-13:
hi, weeks later i had now time to fix it. I got it working finally. With Duet 2 Ethernet and Smart Steppers it works the following way:

Short instructions:
- connect Smart Stepper wires GND, STEP, DIR, EN to Duet external connectors for Duex or LCD Conn (external stepper E2 to E8). You do not need to wire them to the on-board test pins from the TMC2660. You also do not need VIN pin. And you do not need a pullup resistor to force enable signal.
- wire up Smart Stepper with external 24V source
- configure Duet motors with M584 command. Like this (in config.g file)
`; Drives
M569 P5 S1                             ; physical drive 5 goes forwards (E2 connectors)
M569 P6 S1                             ; physical drive 6 goes forwards (E3 connectors)
M569 P7 S1                             ; physical drive 7 goes forwards (E4 connectors)
M584 X5 Y6 Z7                        ; set drive mapping`
M569 P5 R1                             ; #set active high for X 
M569 P6 R1                             ; #set active high for Y
M569 P7 R1                             ; #set active high for Z
- reset Duet
- set the following commands to run some axis
`
G91                  ;rel coords
M564 S0 H0     ; disable bounds
G1 X50             ; move X axis
G1 Y50             ; move Y axis
G1 Z50             ; move Z axis
`

Notes: my printer is still not finished and not running. I will see if the pullup 1k is required maybe but i think RRF fixed with the issue. Please have a look at https://forum.duet3d.com/topic/11492/smart-steppers-conn_lcd/13 which seems to point to that issue (?)

regards and thanks four your efforts. Would be happy if you could update your documentation sources.

regards, Mario

vmario89, 2020-06-07:
to advance the provided information: stumpled upon some post of DC42 at Duet forum some days ago. maybe helpful too > https://forum.duet3d.com/topic/7952/smart-stepper/4

closed 2020-06-07 by vmario89

dc42, 2020-06-07:
> Since the Duet creates additional pulses on the enable line, it is possible that the smart stepper may not produce movement if the pulse is not seen correctly. This could result in a loss of print [...]

The Duet treats the enable line in the standard way, i.e. sets it active the first time the driver is used, and it sets it inactive if you use M18 or M84 to set the motors idle.

The default enable polarity is active low, which is standard for stepsticks and all stepper driver chips that I am familiar with. You can change it to be active high using the R1 parameter of the M569 command.