Misfittech/nano_stepper #41: New Firmware 0.40 direction not work

billydragon, 2019-12-17:
I have BIGTREETECH-S42A-V1.0 motor and driver. working on 0.38 firmware. To make it work i need to change some pins in board.h. With the same config pins on FW 0.39-0.40 do not work any more. Every things fine, LCD work, motor working only one way, can't change direction.
here is board.h file
[board.zip](https://github.com/Misfittech/nano_stepper/files/3971222/board.zip)

coolio986, 2020-01-02:
@billydragon I will check into this and see what changed.

coolio986, 2020-01-06:
@billydragon It appears that you are using the defines for the mechaduino. The pin definitions have not changed from 0.38 to 0.40. It appears the BTT S42A is using pin 2 instead of pin 1 in their board.h. I can make firmware revision to add this functionality.

billydragon, 2020-01-07:
Thank for you information. So what next step i need to change for 0.40 FW ?

coolio986, 2020-01-07:
@billydragon I need to confirm the changes with BTT and add that to FW 0.40. Nothing needs to be done on your part.

coolio986, 2020-01-09:
@billydragon, I have attached a new board.h file. Please use the latest 0.40 FW, and overwrite with this attachment.
[board.zip](https://github.com/Misfittech/nano_stepper/files/4042466/board.zip)