Misfittech/nano_stepper #42: Open Source License

gregsaun, 2020-01-05:
Hi Misfittech team, congrats for your very nice work. I was looking at your GitHub and saw this mention about the firmware license:
> The firmware is licensed as GPL V3 for non-commercial use

GPLv3 explicitly says that there is no commercial limit. This mention makes your license kind of invalid. The other problem is that your project can't really be forked/reused/improved because of that non-commercial mention.

If you really want to have a non-commercial restriction you should look at CC licenses but they are more for art stuffs (they are not very explicit on sharing the source code for example). GPLv3 is for sure the best choice if you really want to make your project open source.

I think you have a very nice project here and it would be good to have a great license coming it with. This would for sure help to grow.

coolio986, 2020-01-06:
@trampas Can you review this?

xerootg, 2021-07-24:
There are many projects with this exact model - dual licensing. Trampas has chosen to extend GPLv3 to non-commercial uses, and not for commercial uses. If your a non-commercial user, then GPLV3 applies to you, but if your not, then GPLv3 does not apply to you and the license of Trampas' choosing does.

gregsaun, 2021-07-25:
It is not possible to extend GPLv3 to non-commercial uses, you have to use another license for that.

Section 7 of https://www.gnu.org/licenses/gpl-3.0.en.html
_All other non-permissive additional terms are considered “further restrictions” within the meaning of section 10. If the Program as you received it, or any part of it, contains a notice stating that it is governed by this License along with a term that is a further restriction, you may remove that term._