Misfittech/nano_stepper #47: How to change vendor/product I or add iSerial

vmario89, 2020-02-07:
Hi,
because i want to have control over each smart stepper by usb connected to a usb hub, i figured out that all Smart steppers have same product id and vendor id. I'd like to persist named devices in raspbian by mapping their id's as fixed rule. Unfortunately that looks not so nice because all IDs of nano zero are the same. How can i change that? i did not find a fitting line of code to adjust that.
[image]

coolio986, 2020-02-10:
You are looking for "boards.txt" in 

<Username>\AppData\Local\Arduino15\packages\misfittech\hardware\samd\1.0.0

`nano_zero.vid.0=0x1209
nano_zero.pid.0=0x8087
nano_zero.vid.1=0x1209
nano_zero.pid.1=0x8086`

vmario89, 2020-02-11:
thanks for the information. This helps a lot! regards, Mario

closed 2020-02-11 by vmario89