Misfittech/nano_stepper #48: ERROR 'Serial5' was not declared in this scope

DanTheHero, 2020-02-07:
Hi, I followed http://misfittech.net/blog/arduino-package-install/ and have a MKS SERVO42A, I bought it without knowing the story about this copy-product, but I need to buy some others soon for my bachelor project. I'm trying the .ino in this hub, the last version but I got this error: 

> In file included from C:\Users\ATLAS\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21.h:69:0,
> 
>                  from C:\Users\ATLAS\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd.h:105,
> 
>                  from C:\Users\ATLAS\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/sam.h:540,
> 
>                  from C:\Users\ATLAS\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.4\cores\arduino/Arduino.h:48,
> 
>                  from sketch\syslog.h:33,
> 
>                  from sketch\stepper_controller.h:27,
> 
>                  from sketch\stepper_controller.cpp:24:
> 
> C:\Users\ATLAS\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.2.0/CMSIS/Device/ATMEL/samd21/include/samd21g18a.h:226:0: warning: "LITTLE_ENDIAN" redefined
> 
>  #define LITTLE_ENDIAN          1
> 
>  
> 
> In file included from c:\users\atlas\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\sys\types.h:67:0,
> 
>                  from c:\users\atlas\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\stdio.h:61,
> 
>                  from sketch\syslog.h:32,
> 
>                  from sketch\stepper_controller.h:27,
> 
>                  from sketch\stepper_controller.cpp:24:
> 
> c:\users\atlas\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4\arm-none-eabi\include\machine\endian.h:17:0: note: this is the location of the previous definition
> 
>  #define LITTLE_ENDIAN _LITTLE_ENDIAN
> 
>  
> 
> sketch\nzs.cpp: In member function 'void NZS::begin()':
> 
> nzs.cpp:639:2: error: 'Serial5' was not declared in this scope
> 
>   Serial5.begin(SERIAL_BAUD);
> 
>   ^~~~~~~
> 
> sketch\nzs.cpp:639:2: note: suggested alternative: 'Serial_'
> 
>   Serial5.begin(SERIAL_BAUD);
> 
>   ^~~~~~~
> 
>   Serial_
> 
> sketch\commands.cpp: In function 'uint8_t kbhit_hw()':
> 
> commands.cpp:1523:9: error: 'Serial5' was not declared in this scope
> 
>   return Serial5.available();
> 
>          ^~~~~~~
> 
> sketch\commands.cpp:1523:9: note: suggested alternative: 'Serial_'
> 
>   return Serial5.available();
> 
>          ^~~~~~~
> 
>          Serial_
> 
> sketch\commands.cpp: In function 'uint8_t getChar_hw()':
> 
> commands.cpp:1528:9: error: 'Serial5' was not declared in this scope
> 
>   return Serial5.read();
> 
>          ^~~~~~~
> 
> sketch\commands.cpp:1528:9: note: suggested alternative: 'Serial_'
> 
>   return Serial5.read();
> 
>          ^~~~~~~
> 
>          Serial_
> 
> sketch\commands.cpp: In function 'uint8_t putch_hw(char)':
> 
> commands.cpp:1532:9: error: 'Serial5' was not declared in this scope
> 
>   return Serial5.write((uint8_t)data);
> 
>          ^~~~~~~
> 
> sketch\commands.cpp:1532:9: note: suggested alternative: 'Serial_'
> 
>   return Serial5.write((uint8_t)data);
> 
>          ^~~~~~~
> 
>          Serial_
> 
> sketch\commands.cpp: In function 'int commandsProcess()':
> 
> sketch\commands.cpp:1579:52: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
> 
>    CommandProcess(&HostUart,Cmds,' ',COMMANDS_PROMPT);
> 
>                                                     ^
> 
> commands.cpp:1589:15: error: 'class Uart' has no member named 'dtr'
> 
>  if (SerialUSB.dtr())
> 
>                ^~~
> 
> sketch\commands.cpp:1591:60: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
> 
>      return CommandProcess(&UsbUart,Cmds,' ',COMMANDS_PROMPT);
> 
>                                                             ^
> 
> exit status 1
> 'Serial5' was not declared in this scope

I really, really hope for help. @trampas @Misfittech Have a great day!

coolio986, 2020-02-08:
Please post your board.h file

jaredgonzales, 2024-01-30:
I hope you found a workaround but am posting in case anyone else is looking back at this.

I encountered this same error when I selected the wrong board. Be sure to install and select the correct board in the Arduino Boards Manager. The name should be `MisfitTech SAMD Boards` in the Boards Manager. When skimming the [instructions](https://misfittech.net/blog/arduino-package-install/) it was easy for me to mistakenly select the "Arduino Zero" board instead of the intended `Nano Zero` board.