Misfittech/nano_stepper #49: MKS SERVO 42 ERROR PIN?

DanTheHero, 2020-02-11:
The MKS servo42 have an pin error? I need to know when the stepper have a certain error in other to interrupt my G-Code, but I don't see an error pin or something.

vmario89, 2020-02-11:
hi,
the docs of Smart Stepper show an error pin which also can be called by USB Serial Interface (e.g. reachable by Linux bash utilizing stty or screen or Windows PuTTY session):
[image]

dont know if Servo42 has the same pins. Maybe ask your question at the mks clone's github repository instead

coolio986, 2020-02-12:
I do not have a MKS servo 42. If you would like one tested, send us one.

closed 2020-02-12 by coolio986