Misfittech/nano_stepper #56: What magnets to use?

vmario89, 2020-06-06:
Hi,
i lost the magnets for some smart steppers time ago and wanted to ask what to buy to replace the lost ones. At the other smart stepper motors there are some 6.35 x 3.175 mm neodyms used and they work fine. Those are the ones which were delivered together with the boards.

I tried some other magnets like 4x2 mm but i think they do not have the correct field strength. My smart steppers only produce horrible error on calibration all the time. No chance to get them quiet. Any recommendations for this?

wjsteele, 2020-06-06:
You need diametric magnets... not axial magnets.

See here for some: https://www.kjmagnetics.com/products.asp?cat=1&scri=14

Bill

vmario89, 2020-06-07:
Hey. Thank you really much for this tip. It's so easy but i just did not came up with that idea :-D I had a look into encoder documentation of the chip and you are totally right. They say "Reference magnet: N35H, 8mm diameter; 3mm thickness."

found that drawing which shows a diametral one
[image]

closed 2020-06-07 by vmario89