Misfittech/nano_stepper #57: Convert to TMC2208?

Shaische, 2020-08-19:
Looking to convert this to a TMC2208. I was wondering if any firmware modifications would be necessary? Any help appreciated!

kb-, 2020-08-19:
The same answer would apply here. https://github.com/Misfittech/nano_stepper/issues/55
Major modifications. The firmware controls a H-bridge with 4 logical inputs. It doesn't generate STEP/DIR for a TMC2208 or alike.

xhejtman, 2020-11-29:
So no plans to add better driver currently?

Shaische, 2020-12-01:
The TMC5160 has encoder support. Page 10 https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC5160A_Datasheet_Rev1.14.pdf
The breakoutboard is open source. https://www.digikey.ca/reference-designs/en/motor-control/motor-control/2592
Download link https://www.trinamic.com/fileadmin/assets/Products/Eval_Drawings/TMC5160-BOB_CAD_Files_V1.2.zip

Maybe it's possible to incorporate TMC5160? I'm not sure if the same setup would work with the other ones like TMC2208?  Sadly above my skill level to incorporate this.